Rickedb/OpenProtocolInterpreter #3: can this open protocol interpreter used on RAD controller?

techengshenfeizi, 2019-01-18:
hi everyone:
  this is not an issue and we just didn't find a right place to post this question so we can only write it down here.

  We are using RAD controller E-RAD BLU 8000-m. did anyone applied this open protocol interpreter to this model? We already consulted RAD and they have confirmed this model supports open protocol but so far we didn't have any luck to use this interpreter to get information from controller.
  
  any suggestion is highly appreciated.

Rickedb, 2019-01-18:
Hi! 
The library does not exactly communicates with the controller itself, it just parses an ASCII string that you provide.
I recommend you to take a look at the sample (if you've not seen yet): https://github.com/Rickedb/OpenProtocolInterpreter/tree/master/sample

techengshenfeizi, 2019-01-22:
again, this is not an issue or bug and we are just asking where open protocol works on ERAD-8000. Most recent news is that we can use sample program provided here to connect and subscribe tightening information from E-RAD 8000. We will keep you posted to see if we can get tightening information.

thank you,

Rickedb, 2019-01-22:
Usually, if controller can "talk open protocol" it will probably work for you.
Any doubts you can ask here or send me an email

gfcittolin, 2019-01-22:
If C# is not mandatory in this case, there's a full implementation of OpenProtocol in NodeJS [node-open-protocol](https://github.com/netsmarttech/node-open-protocol), including connection negotiation and MID parsing/serializing. 
There's then also a [Node-RED node](https://flows.nodered.org/node/node-red-contrib-open-protocol) for it, so you can try-out without much programming

techengshenfeizi, 2019-01-29:
good news!
for ERAD BLU 8000m, we are able to use open protocol implementation here to retrieve MID61 information.

closed 2019-01-29 by techengshenfeizi