Rickedb/OpenProtocolInterpreter #4: The type or namespace name 'MID_0005' could not be found

exist16, 2019-03-22:
Error: The type or namespace name 'MID_0005' could not be found (are you missing a using directive or an assembly reference?)	OpenProtocolInterpreter.Sample	

any idea to resolve it?

Rickedb, 2019-03-24:
Hello!

I haven't updated the samples to use the 2.0.0 version, sorry for that .

Also, if you want to try the samples you will need to rename every "MID_XXXX" to "MidXXXX", also there are some changes to method's names, but not big deal..

I recommend you to read my article at code project before doing it: https://www.codeproject.com/Articles/1190932/The-Atlas-Copco-Open-Protocol-Interpreter#Version2.0.0

exist16, 2019-03-25:
thank you Rickedb,
I will try immediately..

closed 2019-04-02 by Rickedb