Rickedb/OpenProtocolInterpreter #7: ProcessHeader(string package) - Default revision

yanick76, 2019-06-20:
Some messages from some controllers doesn't contains revision, for example MID 0015:
`00420015____________0022017-10-03:14:12:30`
In this case message parsing fails because MID revision parsed as 0 and DataField doesn't contains data for revision 0.
Possible fix is to change default revision to 1 (Mid.cs:100)
`Revision = (package.Length >= 11 && !string.IsNullOrWhiteSpace(package.Substring(8, 3))) ? Convert.ToInt32(package.Substring(8, 3)) : 1,`

Rickedb, 2019-06-21:
As I've seen, even when parsing revision equals to 0, it does change it to rev 1 when parsing the "body" of the package, like in `ProcessDataFields()`

```
protected virtual void ProcessDataFields(string package)
        {
            if (!RevisionsByFields.Any())
                return;

            int revision = HeaderData.Revision > 0 ? HeaderData.Revision : 1; //This part
            for (int i = 1; i <= revision; i++)
                ProcessDataFields(RevisionsByFields[i], package);
        }
```

Can you provide a package example that is failing?

yanick76, 2019-06-21:
Example package is in my comment it's MID 0015. I forgot to mention it's necessary to replace underscores (_) with spaces. In comment preview extra spaces were removed so I replaced them with underscores.

Rickedb, 2019-06-21:
I'm fixing it on mid 0015 override. 
Since making changes on MID class will change every Mid behavior (which is more likely to cause problems/bugs for people).

By now, doing this it will help you out with this problem and I will also study the possibility to modify it to every mid.

Rickedb, 2019-06-21:
Already published on nuget gallery, please update it and close the issue once you check it solved!

yanick76, 2019-06-21:
As far as I understand open protocol revision 0 doesn't exist. In this case it's about way to parse received message from controller and message with revision 0 is always necessary to parse based on revision 1 format.
Anyway, your solution is ok :)

yanick76, 2020-09-15:
I found similar issue in MID 0061 (version 3.2.0)
I switched parameters in Pack method by mistake and I substribed to tightening revision 0.
Tightening data as result:
```text
02310061            010109020203XXXXXXXXXXXXXXXXXXXXXXXXX04XXXXXXXXXXXXXXXXXXXXXXXXX050006001070000080000091101111120008501300092014000870150008731600000170013018000051900040202020-09-15:04:43:09212017-03-01:08:43:16222230004720285
```

I fixed it by modify Mid0061.CS like this
```csharp 
        protected override void ProcessDataFields(string package)
        {
            if (HeaderData.Revision == 0 || HeaderData.Revision == 1 || HeaderData.Revision == 999)
            {
                int revision = HeaderData.Revision > 0 ? HeaderData.Revision : 1;
                ProcessDataFields(RevisionsByFields[revision], package);
            }
            else
            (...)
```

yanick76, 2020-09-15:
Offtopic but I don't want to open new issue for typo in enum file :)
You have typo in file TorqueValuesUnit.cs, value number 3 is LBF_LN but correct should be LBF_IN like Pound-force inch

Rickedb, 2020-09-17:
There are some weird controllers out there, usually the default revision is always 1, but they send empty sometimes. 
Anyway, I'm going to think about handling this behaviour in the next versions.
Also, for your code, better use `if(HeaderData.Revision < 1 || HeaderData.Revision == 999)`, better doing 2 operations than 3.

Rickedb, 2020-09-17:
About the enum, I'm going to change it. However it's may cause break changes. 
Thanks for submitting both issues

yanick76, 2020-09-17:
I'm not sure that was probably PF 3000 with controller sw version 10.5 SR4. I think problem was with tightening subscription (MID 0060). I subscribed to tighttening data with revision 0, by mistake :)
And about my code, sorry i was bit tired. I agree your version is better

Rickedb, 2020-09-17:
Makes sense! Anyway, better handle it in the lib too.

Oh, and you don't need to excuse for that, it was just a clue. Also, don't push yourself too much!

ekdahl, 2021-02-08:
Please note that the specification says:
"If the initial MID Revision (revision 1) is required there is three different ways to get it, either send three spaces or 000 or 001."
I.e. there is no revision 0.
1 is the first revision.

Rickedb, 2021-02-14:
@ekdahl and @yanick76, I'm afraid that this might cause more issues than fixing ours. 
I'm sure that there is no rev 0, and that is a fact. But i've seen some controller to cause bugs when you send revision.

For example, I've seen some cases that when you send KeepAlive with revision set like `00209999001_________` they simply cause an exception and kills the connection with integrator, but it works when sending `00209999____________`. 

Also, OpenProtocolInterpreter allows you to set revision 000, so you can send empty spaces instead of the revision number.
But this will have to be set manually.

Rickedb, 2022-06-06:
Enhancement made at version 5.0.0, please check release notes: https://github.com/Rickedb/OpenProtocolInterpreter/releases/tag/5.0.0

closed 2022-06-06 by Rickedb