Rickedb/OpenProtocolInterpreter #8: Tightening Subscribe Response MID0061

storealutes, 2019-07-23:
The MID0061 object through MIDIncome will not update to new data. When I run more than one tightening the tightening information received in the onTighteningReceived function is always the same as the first tightening received. I have looked at the raw data coming in the logs and I can see they are different and the Tightening ID is increasing in the raw data string but the new data is not being parsed into the object.

Rickedb, 2019-07-30:
Sorry for the delay. I will check it out, I might figure out on my own, but can you exemplify the way you are using it?

About the revision, as far as I know, usually depends on controller's firmware, but there is no problem using the library with another revision.

Rickedb, 2019-07-30:
I've checked, maybe there is a problem with MessagesTemplate possibly using the same instance of MID to parse.

If you could provide the way you're using it, it would be awesome. (if is different from the sample)

Rickedb, 2019-07-31:
I could simulate exactly your problem, I'll be taking care of this and release a new version

Rickedb, 2019-08-12:
Solution will be provided in version 3.X.X, please keep an eye here (or on NuGet)

closed 2019-08-12 by Rickedb