Rickedb/OpenProtocolInterpreter #9: PowerMACS need MID0101,MID0102,MID0103

dxlong8, 2019-08-08:
there is no MID0101,MID0102,MID0103

Rickedb, 2019-08-08:
Hello! 
Yes, it is still missing, because I lack of examples of these Mids I could not develop them.

Yet, if possible, I would like to ask you if you can provide some ASCII examples, once done I can integrate them in the lib.

xGusu, 2019-11-25:
Hi 
MID 0103 = MULTI-SPINDLE RESULT UNSUBSCRIBE        	 => 00200103001         
MID 0005 = CMD. ACCEPT                             	 <= 00240005001         0103
MID 0100 = MULTI-SPINDLE RESULT SUBSCRIBE          	 => 00200100001         
MID 0005 = CMD. ACCEPT                             	 <= 00240005001         0100
MID 0030 = JOBS LIST UPLOAD                        	 => 00200030001         
MID 0031 = JOB LIST                                	 <= 00500031001         140102030405060708091516171850
MID 0038 = JOB SELECTION                           	 => 00220038            03
MID 0005 = CMD. ACCEPT                             	 <= 00240005001         0038
MID 0150 = IDENTIFIER DOWNLOAD                     	 => 00710150            BM3GA021119006019872664-010000012927035403932102201
MID 0005 = CMD. ACCEPT                             	 <= 00240005001         0150
MID 0101 = MULTI-SPINDLE RESULT DATA               	 <= 01920101001 01      010102BM3GA02111900601         030304003050001060001071080006800900092010000800110000012000151300000142019-11-14:14:08:05152019-11-25:11:22:41160091317118010111000809100000
MID 0102 = MULTI-SPINDLE RESULT ACK                	 => 00200102 

Later today if i have time i will add a second spindle to get the result from multiple spindle .
I hope this is a helpfull info for you .

Rickedb, 2019-11-25:
Thank you @xGusu !

Rickedb, 2020-02-12:
Version 3.1.0 containing mids 100, 101, 102 and 103 published and available at NuGet!

closed 2020-02-12 by Rickedb