Rickedb/OpenProtocolInterpreter #10: Conexão C# com power mac

DevilK89, 2019-09-18:
Boa tarde, será que você poderia me dar um auxilio?
Como faço para dar um keepAlive?
baixei a biblioteca, mas mesmo assim ela não reconhece!

Obrigado desde de já.

Rickedb, 2019-09-19:
Ola DevilK89!

Depende da sua conexão, acredito que seja via ethernet, nesse caso você precisa fazer a conexão via socket da forma que você desejar.
Essa biblioteca te ajuda a fazer todo o parsing dos pacotes somente.

No seu caso para mandar um keep alive fica assim:
`var package = new Mid9999().PackBytes(); //Isto gera seu pacote em bytes`

dai basicamente você envia esses bytes através do seu socket.
Se necessário, da uma olhada nos samples aqui: https://github.com/Rickedb/OpenProtocolInterpreter/tree/master/sample

DevilK89, 2019-09-19:
Então, consegui verificar o KeepAlive, usando o mesmo exemplo que você me passou, usando o timer tick, porém agora não consigo conectar no meu power mac, ocorre o seguinte erro; 
System.Net.Sockets.SocketException: 'Uma tentativa de conexão falhou porque o componente conectado não respondeu corretamente após um período de tempo ou a conexão estabelecida falhou! Estou usando o cabo ethernet e configurei o aparelho com a rede, mas mesmo assim ele não conecta!

Rickedb, 2019-09-19:
Normalmente isso é por conta do seu PowerMacs, os que tive oportunidade de trabalhar, já notei que há diversos cuidados em relação a configuração de rede/conexão, mas da própria power macs.
Talvez a porta que ele está ouvindo seja outra também e não a padrão de Power Focus.

Se não me engano, o pessoal da Atlas Copco possui um programinha deles mesmo que consegue conectar e fazer toda essa interface de enviar job e etc. É que agora não me recordo o nome, mas se achar talvez possa te ajudar a diagnosticar o problema.

DevilK89, 2019-09-19:
Consegui conectar, era erro meu na hora de declarar o IP, me ajudou
bastante.

closed 2019-09-19 by Rickedb