Rickedb/OpenProtocolInterpreter #13: sample project job info subscribe output update problem.

exist16, 2019-11-17:
the codes below works only for the first arrived data. it shows only the data of the first package. code ignores consecutive Mid0035 packets although message arrived updated on the output screen.
Which part do I have to modify?

Console.WriteLine($@"JOB INFO RECEIVED (MID 0035): 
                                 Job ID: <{jobInfo.JobId}>
                                 Job Status: <{(int)jobInfo.JobStatus}> ({jobInfo.JobStatus.ToString()})
                                 Job Batch Mode: <{(int)jobInfo.JobBatchMode}> ({jobInfo.JobBatchMode.ToString()})
                                 Job Batch Size: <{jobInfo.JobBatchSize}>
                                 Job Batch Counter: <{jobInfo.JobBatchCounter}>
                                 TimeStamp: <{jobInfo.TimeStamp.ToString("yyyy-MM-dd:HH:mm:ss")}>
");

Sending message: 00200036004         
JOB INFO RECEIVED (MID 0035): 
                                 Job ID: <1>
                                 Job Status: <0> (NOT_COMPLETED)
                                 Job Batch Mode: <0> (ONLY_OK_TIGHTENINGS)
                                 Job Batch Size: <4>
                                 Job Batch Counter: <2>
                                 TimeStamp: <2019-11-17:15:28:05>
Message arrived: 00830035004 0000    010001020030040004050003062019-11-17:15:28:10070010800209011001
Sending message: 00200036004         
JOB INFO RECEIVED (MID 0035): 
                                 Job ID: <1>
                                 Job Status: <0> (NOT_COMPLETED)
                                 Job Batch Mode: <0> (ONLY_OK_TIGHTENINGS)
                                 Job Batch Size: <4>
                                 Job Batch Counter: <2>
                                 TimeStamp: <2019-11-17:15:28:05>

Rickedb, 2019-11-17:
Is this from the sample? 
And which version of the DLL are you working with?

exist16, 2019-11-17:
yes. it is from the sample.
OpenProtocolInterpreter.dll version is 3.0.1

Rickedb, 2019-11-25:
I couldn't simulate it testing the Interpreter directly. Maybe it can be something with the sample.
At first, that makes me relieved.

exist16, 2019-12-07:
I have changed the version of the .dll in the sample with version 3.0.3. that version works well with units.

closed 2019-12-09 by Rickedb