Rickedb/OpenProtocolInterpreter #14: Maybe Bug?

xGusu, 2019-11-22:
After you subscribe to Mid0060, and you receive a a Mid61(tightening result ).
and Keep alive is send before the ACK Mid62 and crash  the communication.
Any tips ? how to solve this  ?

[image]

Rickedb, 2019-11-24:
Hello xGusu, yes, it's a bug. 
That happened because, since the controller sent us a MID0061, it expects to receive a MID0062. 
When we send a keep alive instead of mid 0062 the controller crashes.

My recommendation is to not use the actual Sample project for Production purposes. Still, you're free to use it and modify.

By the way, I'll put this bug on my "ToDo" to fix it! Thank you for opening an issue!

xGusu, 2019-11-25:
I think i also make this bug easy to come because i use DispacherTimer for keep alive . I move it to system timers (like in your example :) ) and the error is not so easy to come , but i think this is still possible to happen if you are very unlucky .
I was thinking to add ManualResetEvent and in keepalive to use WaitOne for extra safety.
Anyway great library .

Rickedb, 2019-11-25:
In another application that we run on production here, we use it as the docs suggest.
We only send keep alives when there is no recent messages, it works well, but since they are asynchronous, you still have the minimal and improbable chance of sending a keep alive when controllers sends a message.

For the sample, I just made it as it is to save time and to have a simple example of how to use the library.

Thank you very much, I appreciate it! Good to know that it's helping you!

Rickedb, 2020-06-24:
Close due to inactivity.

lucciu5, 2021-11-02:
I've added a multi-threading lock, but it didn't help.
Now busy adding a more advanced locking mechanism. In case of a success will do my best to update the sample app.

Thank you for quick reply!

closed 2021-11-10 by Rickedb