Rickedb/OpenProtocolInterpreter #21: NullReference exception when try to set StrategyOptions on MID0061

gurkanindibay, 2020-03-07:
In my project I need to get MID0061 message and save it into the database. To test it  I created and MID0061 object , created a StrategyOptions parameter and set it to Strategy option field of MID0061.
I got a NullReferenceException on SetRawValue which takes byte[] parameter type as value on the line of Size = Value.Length. Because only RawValue is set in this method, Value is everytime null in this case.
```csharp
 public virtual void SetRawValue<T>(Func<char, int, PaddingOrientations, T, byte[]> converter, T value)
        {
            CachedValue = null;
            RawValue = converter(_paddingChar, Size, _paddingOrientation, value);
            Size = Value.Length;
        }

```
I want to fix it but it is being used by more than 100 places. Could anyone help me with this issue?

Thanks

Rickedb, 2020-03-08:
Can you provide me the ASCII package that you are receiving from controller?

So I can simulate what might be happening here to fix it.

gurkanindibay, 2020-03-08:
I encountered the error while writing unit tests. I haven't connected a device yet.
the only thing I did was creating a Mid00061instance and by using new and used 
````csharp
            StrategyOptions strategyOptions = new StrategyOptions();
            strategyOptions.Torque = true;
            strategyOptions.Angle = true;
            strategyOptions.Batch = true;
            strategyOptions.PvtMonitoring = true;
            strategyOptions.PvtCompensate = true;
            strategyOptions.Selftap = true;
            strategyOptions.Rundown = true;
            strategyOptions.CM = true;
            strategyOptions.DsControl = true;
            strategyOptions.ClickWrench = true;
            strategyOptions.RbwMonitoring = true;

            Mid0061 mid = new Mid0061();
            mid.StrategyOptions = strategyOptions;//exception appeared on this line

Rickedb, 2020-03-08:
Thanks for submitting this issue, I've debugged it and you can change from `Value` property to `RawValue` property without any problem, this method (`SetRawValue`) is only used by properties that new the information ONLY in byte[] and cannot be expressed in ASCII values, which it would use the generic `SetValue` method.

Thanks again for submitting it and I will this fix it and deploy a new version of it!

closed 2020-03-09 by Rickedb