Rickedb/OpenProtocolInterpreter #24: (Mid0061) Tighten Error Status not parsing properly

JFlor27, 2020-06-01:
Hello, 

With the following message:
03850061002     0000010000020003SU_ST7.1_ETV100          04                         050000060010702080006709000010000011012213014015016117118119120**0000000002**2100340022004600230040002400069525000252600600270000028000002900360300300031000503200033150340003500000036999900370000003800000039000000400000004100002341084200000430000044      C0720021452020-06-01:15:48:27462020-05-12:07:34:57

Should return **Rundown angle min shut off**, nevertheless, allways return **null**.

Rickedb, 2020-06-24:
Hello. Sorry for the delay.

Going to check what is causing this behaviour

Raykler, 2020-06-25:
The TighteningErrorStatus only "works" if you parse the message as byte[].
For example, for the following message (replace _ for space):
`03850061002_____0000010000020003RA_ST6.2_ETV100__________04_________________________05000006001070208000670900001000001101221301401501611711811912000000081942100340022004600230040002400050525000202600420270000028000002900100300085031000043200033150340003500000036999900370000003800000039000000400000004100001848874200000430000044______C0761275452020-06-25:01:04:39462020-06-24:10:48:53`

I did the following:
```
var mid = _midInterpreter.Parse(Encoding.ASCII.GetBytes(Package));
var m = mid as Mid0061;
```
I received 8 Errors, and also some variables are opposite:
[image]
Nevertheless, I developed the following piece of code because it didn't coincided with the exported data from ToolsTalk, and returns only 2 errors (the same errors that ToolsTalk report, **Trigger lost**, **Rundown angle min shut off**).
```
public static List<string> GetTighteningErrorStatus(string message)
{
    List<string> Errors = new List<string>();

    if (string.IsNullOrEmpty(message))
        return null;
    if (message.Length < 0385)
        return null;

    int.TryParse(message.Substring(147, 10), out int ErrorCode);
    string BinaryData = Convert.ToString(ErrorCode, 2).ToString();
    BinaryData = BinaryData.PadLeft(32, '0');
    int Counter = 33;
    foreach (char c in BinaryData.ToCharArray())
    {
        Counter--;
        if (c == '1')
        {
            switch (Counter)
            {
                case 1: Errors.Add("Rundown angle max shut off"); break;
                case 2: Errors.Add("Rundown angle min shut off"); break;
                case 3: Errors.Add("Torque max shut off"); break;
                case 4: Errors.Add("Angle max shut off"); break;
                case 5: Errors.Add("Self-tap torque max shut off"); break;
                case 6: Errors.Add("Self-tap torque min shut off"); break;
                case 7: Errors.Add("Prevail torque max shut off"); break;
                case 8: Errors.Add("Prevail torque min shut off"); break;
                case 9: Errors.Add("Prevail torque compensate overflow"); break;
                case 10: Errors.Add("Current monitoring max shut off"); break;
                case 11: Errors.Add("Post view torque min torque shut off"); break;
                case 12: Errors.Add("Post view torque max torque shut off"); break;
                case 13: Errors.Add("Post view torque Angle too small"); break;
                case 14: Errors.Add("Trigger lost"); break;
                case 15: Errors.Add("Torque less than target"); break;
                case 16: Errors.Add("Tool hot"); break;
                case 17: Errors.Add("Multistage abort"); break;
                case 18: Errors.Add("Rehit"); break;
                case 19: Errors.Add("DS measure failed"); break;
                case 20: Errors.Add("Current limit reached"); break;
                case 21: Errors.Add("End Time out shutoff"); break;
                case 22: Errors.Add("Remove fastener limit exceeded"); break;
                case 23: Errors.Add("Disable drive"); break;
                case 24: Errors.Add("Transducer lost"); break;
                case 25: Errors.Add("Transducer shorted"); break;
                case 26: Errors.Add("Transducer corrupt"); break;
                case 27: Errors.Add("Sync timeout"); break;
                case 28: Errors.Add("Dynamic current monitoring min"); break;
                case 29: Errors.Add("Dynamic current monitoring max"); break;
                case 30: Errors.Add("Angle max monitor"); break;
                case 31: Errors.Add("Yield nut off"); break;
                case 32: Errors.Add("Yield too few samples"); break;
            }
        }
    }
    return Errors;
}
```

In this example, the BinaryData for code 8194 would be 00000000000000000010000000000010.

I supose that for TighteningErrorStatus2, should work also properly

Rickedb, 2020-06-25:
As @Raykler said, `Parse(string message)` does not work properly for string messages in revisions higher than 1 for `Mid0061`.

This was done because I could not test properly with ASCII Strings and I couldn't go furthermore in the feature, but had certainty on parsing `Bytes[]`.

Now with some time, I plan to work on it, also @Raykler provided good example for doing it.

Raykler, 2020-11-02:
Have you been able to test the issue?
I would like to update to the last version.

Rickedb, 2020-11-09:
@Raykler Sorry for the delay! Been in a rush at work since the beginning of covid. I'm working on it this week, hope to have it until thursday.

Rickedb, 2020-11-10:
Solved in version 3.2.2, please download on NuGet!

closed 2020-11-10 by Rickedb