Rickedb/OpenProtocolInterpreter #25: Mid0106 error

Raykler, 2020-06-22:
The following piece of code in PowerMACS/Mid0106 is wrong.
```
public bool SimpleStatus
{
    get => GetField(1, (int)DataFields.MODE_NUMBER).GetValue(_boolConverter.Convert);
    set => GetField(1, (int)DataFields.MODE_NUMBER).SetValue(_boolConverter.Convert, value);
}
```

Should be:
```
public bool SimpleStatus
{
    get => GetField(1, (int)DataFields.SIMPLE_STATUS).GetValue(_boolConverter.Convert);
    set => GetField(1, (int)DataFields.SIMPLE_STATUS).SetValue(_boolConverter.Convert, value);
}
```

Rickedb, 2020-06-24:
Thanks for providing it, must have missed it when copy pasting.
Going to fix it for next version!

Rickedb, 2020-06-26:
Version 3.2.0 published with fixes

closed 2020-06-26 by Rickedb