Rickedb/OpenProtocolInterpreter #26: DecimalConverter bug

Raykler, 2020-06-25:
Due to some reason, when you convert BoltData from MID0106 sometimes the data is not well converted (for example when the tool loosens and returns a negative angle).
```
public override decimal Convert(string value)
{
    decimal decimalValue = 0;
    if (value != null)
        decimal.TryParse(value.ToString(), out decimalValue);

    return decimalValue;
}
```

I fixed it adding the following replace.
```
public override decimal Convert(string value)
{
    decimal decimalValue = 0;
    if (value != null)
        decimal.TryParse(value.ToString().Replace('.', ','), out decimalValue);

    return decimalValue;
}
```

Perhaps can be also fixed using CultureInfo, the language of my client is es-ES.

Rickedb, 2020-06-25:
It's not common to see decimal data with commas, but since there's the possibility of having commas instead of dots It's cool to handle both cases.

Going to implement it on next version. Thanks for providing info and code.

Rickedb, 2020-06-25:
@Raykler Tried to emulate a negative angle but without success, do you have any example of it?

Raykler, 2020-06-26:
Here you have, for Mid0106 Rev.1, replace the _ with space.

`03370106____________01030201030000200614040105RA_ST5.1_PwMACS_____062020-06-26:09:07:10075008Mode_50_____________09010211________________________________________1202130114015_16117_______18-573.0819_______20_______21_______22_______130214015_16117_______18-574.3019_______20_______21_______22_______2301Data_No_Station_____I_100000004009`

Raykler, 2020-11-02:
Have you been able to test the issue?
I would like to update to the last version.

Rickedb, 2020-11-10:
Solved in version 3.2.2, please download on NuGet!

closed 2020-11-10 by Rickedb