Rickedb/OpenProtocolInterpreter #27: Mid0076 Rev.2

Raykler, 2020-06-25:
I added the Revision 2 for the alarm Mid0076.

It's in the specification of the release 2.8.0 that I found here:
https://github.com/netsmarttech/node-open-protocol/blob/master/docs/OpenProtocolSpecification_R280.pdf

`private const int LAST_REVISION = 2;`

```
protected override Dictionary<int, List<DataField>> RegisterDatafields()
{
    return new Dictionary<int, List<DataField>>()
    {
        {
            1, new List<DataField>()
                    {
                        new DataField((int)DataFields.ALARM_STATUS, 20, 1),
                        new DataField((int)DataFields.ERROR_CODE, 23, 4, ' ', DataField.PaddingOrientations.LEFT_PADDED),
                        new DataField((int)DataFields.CONTROLLER_READY_STATUS, 29, 1),
                        new DataField((int)DataFields.TOOL_READY_STATUS, 32, 1),
                        new DataField((int)DataFields.TIME, 35, 19)
                    }
        },
        {
            2, new List<DataField>()
                    {
                        new DataField((int)DataFields.ALARM_STATUS, 20, 1),
                        new DataField((int)DataFields.ERROR_CODE, 23, 4, ' ', DataField.PaddingOrientations.LEFT_PADDED),
                        new DataField((int)DataFields.CONTROLLER_READY_STATUS, 30, 1),
                        new DataField((int)DataFields.TOOL_READY_STATUS, 33, 1),
                        new DataField((int)DataFields.TIME, 36, 19)
                    }
        },

    };
}
```

Rickedb, 2020-06-25:
Also checked the Version 2.10.0 docs. There is revision 3 for Mid 0076.

I'll provide it in the next version. Thanks for providing code again!

Rickedb, 2020-06-26:
Version 3.2.0 publish with new revisions

closed 2020-06-26 by Rickedb