Rickedb/OpenProtocolInterpreter #33: May be bug or Some other problem

amit-tomar-github, 2020-10-02:
I was communicating to one Open protocol controller with 3 tool. So first we try to use MID0100 but for that we are getting parsing error while receiving the response from the controller. So vendor of controller suggest that we can use 3 different port for 3 tool of the controller so we changed accordingly. We are testing everthing with the sample app in the repo. So we connected successfully but we are getting the response from all the 3 tool  multple times. Even controller is sending one time but we are receiving multiple times. This is the first problem we faced but after some time we are having another big problem , controller is sending the data of all the 3 tool but we are receiving only for either 1 or 2 tool. When we restart the app then we again receive the data from all the 3 tool for 2-3 times but again then we received the data only for 1 or 2 tool. Here I am attaching code and log file as well.
[Log_20201001.log](https://github.com/Rickedb/OpenProtocolInterpreter/files/5316539/Log_20201001.log)

[Code.txt](https://github.com/Rickedb/OpenProtocolInterpreter/files/5316534/Code.txt)

yanick76, 2020-10-08:
I think this is off-topic issue because source of this issue is not related to this project.
Anyway, your problem is you are trying to process asynchronous data synchronously. Sample is sufficient for 1 controller but no for 3.

amit-tomar-github, 2020-10-08:
Hi, Thank you for your time and feedback. I will use this code, But I was short on time for this project so I figured out there is MID0100 and MID0101 for multispindle so I am using this MID in my project. But I am having one problem with that sometimes I am getting a message in the log that keep alive not received and then I never got the data again. So what I am doing on keepalive timer I am reconnecting to the controller(calling btnConnect) event. I don't know why this happening. Even though I will change in that code too according to this code then will see how it will work..BY THE WAY, THANK YOU AGAIN

yanick76, 2020-10-08:
You are welcome, let me know result :)
There is bug in MID 0100 and I forgot to report it. Constant LAST_REVISION has value 4 but this revision isn't defined and flag "sendOnlyNewData" wasn't implemented at all. Anyway, "Send only new data" is for revision 3+ and I can test MID 0100 on 2 PF4000 controllers and one PF6000 and each controller can use only MID 0100 with revision 1 so I think this flag is useless (for now).

Here is modified code:
```csharp
using OpenProtocolInterpreter.Converters;
using System.Collections.Generic;

namespace OpenProtocolInterpreter.MultiSpindle
{
    /// <summary>
    /// Multi-spindle result subscribe
    /// <para>
    ///     A subscription for the multi-spindle status. For Power Focus, the subscription must 
    ///     be addressed to a sync Master. 
    /// </para>    
    /// <para>
    ///     This telegram is also used for a PowerMACS 4000 system 
    ///     running a press instead of a spindle. A press system only supports revision 4 and higher 
    ///     of the telegram and will answer with <see cref="Communication.Mid0004"/>, MID revision unsupported if a subscription 
    ///     is made with a lower revision.
    /// </para>
    /// <para>Message sent by: Integrator</para>
    /// <para>
    ///     Answer: <see cref="Communication.Mid0005"/> Command accepted or 
    ///     <see cref="Communication.Mid0004"/> Command error, Controller is not a sync master/station controller, 
    ///     Multi-spindle result subscription already exists or MID revision unsupported
    /// </para>
    /// </summary>
    public class Mid0100 : Mid, IMultiSpindle, IIntegrator
    {
        private readonly IValueConverter<long> _longConverter;
        private readonly IValueConverter<bool> _boolConverter;
        private const int LAST_REVISION = 4;
        public const int MID = 100;

        public long DataNumberSystem
        {
            get => GetField(2, (int)DataFields.DATA_NUMBER_SYSTEM).GetValue(_longConverter.Convert);
            set => GetField(2, (int)DataFields.DATA_NUMBER_SYSTEM).SetValue(_longConverter.Convert, value);
        }

        public bool SendOnlyNewData
        {
            get => GetField(3, (int)DataFields.SEND_ONLY_NEW_DATA).GetValue(_boolConverter.Convert);
            set => GetField(3, (int)DataFields.SEND_ONLY_NEW_DATA).SetValue(_boolConverter.Convert, value);
        }

        public Mid0100() : this(LAST_REVISION)
        {

        }

        public Mid0100(int revision = LAST_REVISION) : base(MID, revision)
        {
            _longConverter = new Int64Converter();
            _boolConverter = new BoolConverter();
        }

        public Mid0100(int revision, bool sendOnlyNewData) : this(revision)
        {
            SendOnlyNewData = sendOnlyNewData;
        }

        public Mid0100(int revision, int dataNumberSysten, bool sendOnlyNewData) : this(revision)
        {
            DataNumberSystem = dataNumberSysten;
            SendOnlyNewData = sendOnlyNewData;
        }

        protected override Dictionary<int, List<DataField>> RegisterDatafields()
        {
            return new Dictionary<int, List<DataField>>()
            {
                { 1, new List<DataField>() },
                {
                    2, new List<DataField>()
                            {
                                new DataField((int)DataFields.DATA_NUMBER_SYSTEM, 20, 10, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                            }
                },
                {
                    3, new List<DataField>()
                            {
                                new DataField((int)DataFields.SEND_ONLY_NEW_DATA, 30, 1, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                            }
                },
                {
                    4, new List<DataField>()
                }
            };
        }

        public enum DataFields
        {
            DATA_NUMBER_SYSTEM,
            SEND_ONLY_NEW_DATA
        }
    }
}
```

amit-tomar-github, 2020-10-11:
Sorry for the late reply and thanks for posting this code. I was busy with this controller thing where I am getting data more than one time even the controller is sending only one time but I am getting some times 3-4 times. I am using Revision 1 for 0100 but now I will test your code too. I have found one more problem with Mid0101 when parsing data then during conversion there is a problem with the substring variable
[image]

yanick76, 2020-10-13:
Yes, I found this bug too but I forgot it in my last comment :)
I fixed it this way (I deleted extra "* 18"):
```csharp
public override IEnumerable<SpindleOrPressStatus> Convert(string value)
        {
            for (int i = 0; i < value.Length; i += 18)
            {
                // var spindleValue = value.Substring(i * 18, 18);
                var spindleValue = value.Substring(i, 18);
                yield return new SpindleOrPressStatus()
                {
                    SpindleOrPressNumber = _intConverter.Convert(spindleValue.Substring(0, 2)),
                    ChannelId = _intConverter.Convert(spindleValue.Substring(2, 2)),
                    OverallStatus = _boolConverter.Convert(spindleValue.Substring(4, 1)),
                    TorqueOrForceStatus = (TighteningValueStatus)_intConverter.Convert(spindleValue.Substring(5,1)),
                    TorqueOrForce = _decimalConverter.Convert(spindleValue.Substring(6, 6)),
                    AngleOrStrokeStatus = _boolConverter.Convert(spindleValue.Substring(12, 1)),
                    AngleOrStroke = _intConverter.Convert(spindleValue.Substring(13,5))
                };
            }
        }
```

amit-tomar-github, 2020-10-13:
Hi,
Thanks for your feedback. Now I am having one strange problem some data is missing. When I checked the log then When I received the tightening data then if before sending confirmation MID, keep alive mid sent then the connection is lost and then I am reconnecting to the controller but till then I have lost the data. This happens mostly in the case of first data NG then connection lost then I only had NG data because reconnection to the controller will take some time and till then the operator has passed the operation to the next process.

yanick76, 2020-10-14:
I think you didn't sent some ACK or didn't sent it in right order. I checked if keepalive (MID 9999) can drop connection and I sent keepalive before ack on purpose and connection was unaffected. I had similar issues when I forgot to send ack. You can find this in Open Protocol specification:

***4.3.3 Subscribed data message acknowledge***
*Depending on the subscription message” No Acknowledge Flag” settings in the message header the integrator is acknowledging or not acknowledging. If the flag is NOT set the integrator shall acknowledge the data event messages by sending the corresponding acknowledge MID, otherwise not. If no acknowledge is received before the response timeout, the controller will re-send the message up to three times. After three attempts the controller will consider the connection as lost.*

You can try to subscribe with NoAck option (byte 12 in message header) to test this possibility. "No ack" option is considered as unreliable so you should restore this option after testing.

amit-tomar-github, 2020-10-14:
I am sending the ACK but I feel so while checking log I noticed I am getting message keep alive not received if just before ACK I am sending keep alive MID and after that code enter to reconnect method which will take some time and till then I have lost some OK data. I am send ACK in the following way
```
  private void OnTighteningReceived3(MIDIncome e)
  {
            Log.Information("Tightening Data Received For Port 9003");
            // lblManualMidResult.Text = "Tightening Data Received";
            driverPort3.SendMessage(e.Mid.BuildAckPackage());
}
```
I have 3 controller for multispindle and one controller with single tool. In both the cases I am having problem..Can you help me what wrong I am doing and where ?  Can controller handle multiple operation like when controller is waiting for some ACK and we are sending MID9999 for keep alive ?

yanick76, 2020-10-16:
I tried to send keepalive before ACK on purpose and this is result:
```
15.10 05:02:59.371 ... DEBUG: READ << 00420015            0042015-05-26:08:56:56
15.10 05:02:59.380 ... >> KeepAliveAtlas >> (00209999            )
15.10 05:02:59.383 ... << Parameter set selected - 4 (2015-05-26 08:56:56)
15.10 05:02:59.385 ... DEBUG: WRITE >> PsetNo_Acknowledge >> (00200016001         )
15.10 05:02:59.408 ... >> PsetNo_Acknowledge >> (00200016001         )
15.10 05:02:59.410 ... Parameter set ID: 4
15.10 05:02:59.428 ... DEBUG: READ << 00209999            
```
So, I received response from controller 48ms later and no connection lost error or something like this.

It's hard to tell where is problem when I have only fragments of your code. I don't see try-catch block so I can guess if you're trying to log different events in same time you can get "file is in use" exception. 
Try to run logger in different thread (if you don't do it already) or try to swap lines like this:
```csharp
private void OnTighteningReceived3(MIDIncome e)
{
            driverPort3.SendMessage(e.Mid.BuildAckPackage());
              // lblManualMidResult.Text = "Tightening Data Received";
            Log.Information("Tightening Data Received For Port 9003");          
}
```

amit-tomar-github, 2020-10-16:
Thanks for your reply.. Yes, i am using the try-catch block and not getting error message. But I will try to change as you have suggested. Thanks again for your help.

yanick76, 2020-10-17:
I tried to modify sample to work with multi-spindle tightening and I found problem is probable in SimpleTcpClient not your code. This TCP client is synchronous, it's good for sample but not for production. I lost connection to controller when I receive tightening result and in same time keepalive MID was sent and client waiting for response.
Check this code:
```csharp
public void WriteLine(string data)
        {
            if (string.IsNullOrEmpty(data)) { return; }

            while (this.waitingForResponse)
                Thread.Sleep(10);
            if (data.LastOrDefault() != Delimiter)
            {
                Write(data);
            }
            else
            {
                Write(data);
            }
        }
```

You can see You can't send ACK MID when other MID was sent with request for response and client is waiting for requested response.

amit-tomar-github, 2020-10-17:
Thanks for your time again..this is exactly happening with me..Now what should I do and where and what should I change to run this code in production?

yanick76, 2020-10-19:
I tried to change TCP class to SimpleTcp [1] (NuGet name is SuperSimpleTcp) and got no connection lost during one and half hour. I want share this multi-spindle sample with this project but I don't when it will be ready.

[1] https://github.com/jchristn/SimpleTcp

xGusu, 2020-10-20:
Hi 
I think i had a similar issue . To improve the process don't send keep alive if you already send another command. Use keep alive only you have nothing to communicate. This will not fix your error but will decrease the chance of getting error.

amit-tomar-github, 2020-10-20:
@xGusu  Thanks for your reply. I tried to do so by creating one flag which will check if the controller is not waiting for ACK then only keepalive will send the data. But as you said it does not fix the error and due to critical data I can not say to the client that you have to manage like this only.
@yanick76  Thanks for sharing the class reference link I will try to use this class and hoping the issue will be resolved.
**Thanks again for your time and help @yanick76 @xGusu**

yanick76, 2020-10-20:
Modified sample is included in different issue
https://github.com/Rickedb/OpenProtocolInterpreter/issues/34

yanick76, 2020-10-21:
I left this sample running for 24 hours and i got no connection lost (and no crash or memory/resources leak) so you can use this sample I think it's stable enough.
I hope this helps you.

amit-tomar-github, 2020-10-21:
Thank you so much for your help & time. I will use this sample in my production application. Thank you again

Rickedb, 2020-11-16:
Closing issue due to inactivity and it seems that was solved. 
If problem persists, just reopen it.

closed 2020-11-16 by Rickedb