Rickedb/OpenProtocolInterpreter #38: Mid0061 - TighteningErrorStatus not working properly

Raykler, 2020-11-11:
OpenProtocolInterpreter Version: 3.2.2

For the following string (replace _ with space):
> 03850061002_____0000010000020003RA_ST5.3_ETV150__________04_________________________05000006002070208000670900001000001101221301401501611711811912000000081942101030022012700230115002400312025000202600200270000028000002900360300850031000023200033150340003500030036999900370000003800030039002000400000004100001495124200000430000044______C0690301452020-11-09:17:35:59462020-07-24:11:35:42`

When decoded sending the telegram as string, detects properly TriggerLost and RundownAngleMinShutOff.
But if you send it as byte[] 

> MidInterpreter _midInterpreter = new MidInterpreter().UseAllMessages();
> // Trigger lost , Rundown angle min shut off
> string telegram = "03850061002     0000010000020003RA_ST5.3_ETV150          04                         05000006002070208000670900001000001101221301401501611711811912000000081942101030022012700230115002400312025000202600200270000028000002900360300850031000023200033150340003500030036999900370000003800030039002000400000004100001495124200000430000044      C0690301452020-11-09:17:35:59462020-07-24:11:35:42";
> var mid = _midInterpreter.Parse(**Encoding.ASCII.GetBytes(telegram)**);
>var m = mid as Mid0061;

The TighteningErrorStatus that detects is:
>     =>AngleMaxMonitor: true
>     AngleMaxShutOff: false
>     CurrentLimitReached: false
>     CurrentMonitoringMaxShutOff: false
>     DisableDrive: false
>     DsMeasureFailed: false
>     =>DynamicCurrentMonitoringMax: true
>     DynamicCurrentMonitoringMin: false
>     =>EndTimeOutShutOff: true
>     MultistageAbort: false
>     =>PostViewTorqueAngleTooSmall: true
>     PostViewTorqueMaxTorqueShutOff: false
>     PostViewTorqueMinTorqueShutOff: false
>     PrevailTorqueCompensateOverflow: false
>     PrevailTorqueMaxShutOff: false
>     PrevailTorqueMinShutOff: false
>     Rehit: false
>     =>RemoveFastenerLimitExceeded: true
>     RundownAngleMaxShutOff: false
>     RundownAngleMinShutOff: false
>     =>SelftapTorqueMaxShutOff: true
>     =>SelftapTorqueMinShutOff: true
>     SyncTimeout: false
>     ToolHot: false
>     TorqueLessThanTarget: false
>     TorqueMaxShutOff: false
>     TransducerCorrupt: false
>     TransducerLost: false
>     TransducerShorted: false
>     =>TriggerLost: true
>     YieldNutOff: false
>     YieldTooFewSamples: false

Rickedb, 2020-11-12:
This is because it parses to byte[] the entire string as ASCII. 

My advice is, if you want to use byte[] method, you must use it raw from what is coming from the TCPClient/Serial. Otherwise, parsing to ASCII will shambles everything. 
Or if you have an ASCII string in your hands parse it with string method.

### Explaining what has gone wrong:

When parsing directly from string, we MUST parse the **TighteningErrorStatus** string to an integer and then parse it to byte[].

In this case we have the integer 8194 which turns to a 4 byte array (0 0 32 2) and must be the same as this: `00000000 00000000 00100000 00000010` 

Each bit there represents a status: **Trigger lost** and **Rundown angle min shut off**.

But when you parse the string "0000008194", it will generate bytes from the string `"0"` not from the integer. Resulting in a way different 10 byte array: `48 48 48 48 48 48 56 49 57 52`. 

> Notice that each 0 turns to 48 when parsed to byte.

Rickedb, 2021-04-02:
Closing issue due to inactivity. Reopen if problem persists.

closed 2021-04-02 by Rickedb