Rickedb/OpenProtocolInterpreter #39: DecimalConvert still not working for ES computer

Raykler, 2020-11-11:
I've  been testing for a while, and I've found a solution for the issue.
The .Replace(',', '.') didn't fix it properly.
The following code works fine in my computer (I assume should work properly in all computers due to the CultureInfo).
        
        public override decimal Convert(string value)
        {
            decimal decimalValue = 0;
            if (value != null)
                decimal.TryParse(value.Replace(',', '.'), NumberStyles.AllowDecimalPoint, new CultureInfo("en-US"), out decimalValue);

            return decimalValue;
        }

I'm not sure if it's necessary change it in other places (I haven't found a DoubleConvert, FloatConverter or other type of decimal)

Rickedb, 2020-11-12:
Probably it's taking the Threads culture when parsing. Going to force en-US for this.

Rickedb, 2020-11-16:
Related to #26

Rickedb, 2020-12-22:
Resolved on version 3.3.0

closed 2020-12-22 by Rickedb