Rickedb/OpenProtocolInterpreter #40: Subscribing to MID 0900 using MID 0008

Josep313, 2020-11-11:
As stated in the OpenProtocol PDF:

> 5.27.1.2 Subscribe, MID 0900 Trace data message
Used by the integrator to subscribe on trace data. MID is used together with MID 0008, see Table 23 MID 0008, revision 1, this begins at byte 30 after the message header.
Message sent by: Integrator
Answer: MID 0005 Command accepted with MID 900 in the data field or MID 0004 Command error with MID 900 in the data field and error code, Subscription MID Revision unsupported or Subscription already exist or Subscription on specific data not supported or Invalid data

I am trying to implement **Mid0900** and by using **Mid0008** to subscribe to Trace Curve Data messages sent by the controller.  However I always get a **Mid0004** with error code 99 (**Unknown Mid**) as a response from the **PF4000 controller.** I am hoping someone may see the error in my logic or this might be a bug. 

```C#           
           // Create the Mid0008 message to be sent to the controller
            var mid0008 = new Mid0008()
            {
               SubscriptionMid = "0900",
               WantedRevision = 001,
               ExtraDataLength = 35,
               ExtraData = "10000000000000000000000000000101002",
            };

            pack = protocolFunctions.SendAndWaitForResponse(mid.Pack(), TimeSpan.FromSeconds(10));
            if (pack != null)
            {
               if (pack.HeaderData.Mid == Mid0004.MID)
               {
                  var mid04 = pack as Mid0004;
                  Console.WriteLine($@"Error (MID 0004):
                                         Error Code: <{mid04.ErrorCode}>
                                         Failed MID: <{mid04.FailedMid}>");
               }
               else
                  Console.WriteLine($"Curve subscription accepted!\n");
            }
            protocolFunctions.AddUpdateOnReceivedCommand(typeof(Mid0900), OnCurveReceived);

```

I believe my extra data logic is correct according to Table 195 Extra data field in the OpenProtocol PDF

| Send Alternatives                            | 1  | UI | Following alternatives are available. One ASCII digit 0=Only new data, 1= Stored data from given index, 2 Stored data from given time stamp, 3 Stored data between two indexes, 4 Stored data between two given time stamps in Unix time ( Seconds since 1970-01-01).. If = 0 then only the new data generated after the subscription is done is sent to the subscriber. Old unsent data will not be sent to the subscriber. If = 1 the data from given INDEX is sent inclusive the latest stored. If = 2 the data from given time stamp in Unix format is sent inclusive the latest stored. If = 3 the data between two given indexes is sent If = 4 the data between two given time stamps in Unix time is sent |
|----------------------------------------------|----|----|-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| Data Identifier Time Stamp type              | 19 | T  | The identifier is a Time stamp of the requested data. The first data sent will be the first data and inclusive this time stamp and forward up to and inclusive the last one. If the data is not found, rewind will be to oldest possible data. All data from this point up to the newest available will be sent directly on subscribe. If not used filled in with zeroes e.g at alternative 1. At alternative 2 it contains the Time Stamp ex. 2015.10.01:19:01:30.                                                                                                                                                                                                                                               |
| Data identifier Index type or unix time type | 10 | UI | The Identifier INDEX or the UNIX time (at Alternative 2) of the data to rewind to. 10 bytes. Only used for old stored process data. The first data sent will be the data from and inclusive this point and forward up to and inclusive the last one. If the data is not found, or if the value is 0, rewind will be to oldest possible data. All data from this point up to the newest available will be sent directly on su                                                                                                                                                                                                                                                                                      |
| Number of trace types                        | 2  | UI | The number of trace types subscribed for                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          |
| Trace Type                                   | 3  | UI | Type of the trace curve subscribed for. This field is repeated the Number of trace types. 1 = Angle trace 2 = Torque trace 3 = Current trace 4 = Gradient trace 5 = Stroke trace 6 = Force trace                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  |

I also added a new MID category called TighteningResults:

```C#

namespace OpenProtocolInterpreter.TighteningResults
{
   internal class TighteningResultMessages : MessagesTemplate
   {
      public TighteningResultMessages() : base()
      {
         _templates = new Dictionary<int, MidCompiledInstance>()
            {
                { Mid0900.MID, new MidCompiledInstance(typeof(Mid0900)) },
               // { Mid0901.MID, new MidCompiledInstance(typeof(Mid0901)) },
            };
      }

      public TighteningResultMessages(IEnumerable<Type> selectedMids) : this()
      {
         FilterSelectedMids(selectedMids);
      }

      public TighteningResultMessages(InterpreterMode mode) : this()
      {
         FilterSelectedMids(mode);
      }

      public override bool IsAssignableTo(int mid) => mid > 900 && mid < 902;
   }
}

```

Which is then added _MessageInterpreterExtensions.cs_:

```C#

public static MidInterpreter UseAllMessages(this MidInterpreter midInterpreter, InterpreterMode mode = InterpreterMode.Both)
        {
            return midInterpreter
                .UseAlarmMessages(mode)
                .UseApplicationControllerMessage(mode)
                .UseApplicationSelectorMessages(mode)
                .UseApplicationToolLocationSystemMessages(mode)
                .UseAutomaticManualModeMessages(mode)
                .UseCommunicationMessages(mode)
                .UseIOInterfaceMessages(mode)
                .UseJobMessages(mode)
                .UseAdvancedJobMessages(mode)
                .UseMotorTuningMessages(mode)
                .UseMultipleIdentifiersMessages(mode)
                .UseMultiSpindleMessages(mode)
                .UseOpenProtocolCommandsDisabledMessages(mode)
                .UseParameterSetMessages(mode)
                .UsePLCUserDataMessages(mode)
                .UsePowerMACSMessages(mode)
                .UseResultMessages(mode)
                .UseStatisticMessages(mode)
                .UseTighteningMessages(mode)
                .UseTighteningResultsMessages(mode) // added here
                .UseTimeMessages(mode)
                .UseToolMessages(mode)
                .UseUserInterfaceMessages(mode)
                .UseVinMessages(mode);
        }

```

Am I missing something in regards to subscripting to **Mid0900** using **Mid0008**? Does the **PF4000** use **MID0900**?

Insight on this would be greatly appreciated

Josep313, 2020-11-11:
Added **Mid0900.cs** if relevant

Not sure how _VariableDataFields_ are indexed inside the _RegisterDataFields()_ method
```C#
using System;
using System.Collections.Generic;
using OpenProtocolInterpreter.Converters;
using OpenProtocolInterpreter.Result;
using System.Linq;

namespace OpenProtocolInterpreter.TighteningResults
{
   public class Mid0900 : Mid, ITighteningResults, IController
   {
      private readonly IValueConverter<int> _intConverter;
      private readonly IValueConverter<IEnumerable<VariableDataField>> _variableDataFieldListConverter;
      private readonly IValueConverter<DateTime> _dateConverter;

      private const int LAST_REVISION = 1;
      public const int MID = 900;

      public Mid0900() : base(MID, LAST_REVISION)
      {
         _intConverter = new Int32Converter();
         _dateConverter = new DateConverter();
         _variableDataFieldListConverter = new VariableDataFieldListConverter(_intConverter);
         VariableDataFields = new List<VariableDataField>();

         //NullChar = 0x00;

      }

      protected override Dictionary<int, List<DataField>> RegisterDatafields()
      {
         return new Dictionary<int, List<DataField>>()
            {
                {
                    1, new List<DataField>()
                    {
                        //new DataField((int) DataFields.RESULT_DATA_IDENTIFIER, 20, 10, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                        //new DataField((int) DataFields.TIME_STAMP, 30, 19, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                        //new DataField((int) DataFields.NUMBER_OF_PIDS, 39, 3, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                        //new DataField((int)DataFields.NUMBER_DATA_FIELDS, 40, 3, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                        //new DataField((int) DataFields.TRACE_TYPE, 43, 2, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                        //new DataField((int) DataFields.TRANSDUCER_TYPE, 45, 2, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                        //new DataField((int) DataFields.Unit, 47, 3, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                        //new DataField((int) DataFields.NUMBER_OF_PIDS, 50, 2, '0', DataField.PaddingOrientations.LEFT_PADDED, false),
                        //new DataField((int)DataFields.RESOLUTION_FIELDS, 100000, 0, false), //defined at runtime
                        //new DataField((int)DataFields.VARIABLE_DATA_FIELDS, 100000, 0, false) //defined at runtime
                        new DataField((int) DataFields.RESULT_DATA_IDENTIFIER, 20, 10),
                        new DataField((int) DataFields.TIME_STAMP, 30, 19),
                        new DataField((int) DataFields.NUMBER_OF_PIDS, 39, 3),
                        new DataField((int)DataFields.NUMBER_DATA_FIELDS, 40, 3),
                        new DataField((int) DataFields.TRACE_TYPE, 43, 2),
                        new DataField((int) DataFields.TRANSDUCER_TYPE, 45, 2),
                        new DataField((int) DataFields.Unit, 47, 3),
                        new DataField((int) DataFields.NUMBER_OF_PIDS, 0, 3),
                        new DataField((int)DataFields.VARIABLE_DATA_FIELDS, 0, 0, false), //defined at runtime
                        new DataField((int)DataFields.RESOLUTION_FIELDS, 0, 0, false), //defined at runtime
                        

                    }
                }
            };
      }

      public int ResultDataIdentifier
      {
         get => GetField(1, (int)DataFields.RESULT_DATA_IDENTIFIER).GetValue(_intConverter.Convert);
         set => GetField(1, (int)DataFields.RESULT_DATA_IDENTIFIER).SetValue(_intConverter.Convert, value);
[... 81 lines cut ...]
      public enum DataFields
      {
         //rev 1
         RESULT_DATA_IDENTIFIER = 0,
         TIME_STAMP = 1,
         NUMBER_OF_PIDS = 2,
         TRACE_TYPE = 3,
         TRANSDUCER_TYPE = 4,
         Unit = 5,
         NUMBER_DATA_FIELDS = 6, // data field repeats?
         VARIABLE_DATA_FIELDS = 7,
         NUMBER_OF_RESOLUTION_FIELDS = 8,
         RESOLUTION_FIELDS = 9,
         NUMBER_OF_TRACE_SAMPLES = 10,
         NUL_CHAR = 11, // 0x00 is sent here
         TRACE_SAMPLE = 12,
         // rev 2
         REQUEST_MID = 13,

      }

   }
}

```

Rickedb, 2020-11-12:
At first sight, the problem seems to be with controller.
If you're in doubt if you're generating the correct package, what you can do is to force the string you want to send and see what will be the controller's response.

In the open protocol documentation they have exactly what you need as an example, it might help you: `006800080010________0900001390_____________________________020010020`

If the response is **Unknown Mid**, the controller probably doesn't accepts mid 0008 or maybe does not recognize mid 0900.
Since I haven't worked with those in Power Focus, I cannot ensure that this is the case, but it can be.

In addition, what I recommend you to do is to check if it's firmware is updated to the latest.

Josep313, 2020-11-24:
After some digging, I found out the in order to use MIDs: 900, 901, 2500, 2505, the controller must have the firmware version of W14.x

closed 2020-11-24 by Josep313