Rickedb/OpenProtocolInterpreter #44: Keepalive gets revision 0

ekdahl, 2021-02-08:
I'm using the library for a MID interpreter. Entering a keepalive "00209999001         " gets reported as revision 0 where it is in fact 1.

See https://github.com/Rickedb/OpenProtocolInterpreter/blob/master/src/OpenProtocolInterpreter/KeepAlive/Mid9999.cs#L27
I guess that should be 1 instead as there does not exist revision 0 for any MID.

Rickedb, 2021-02-10:
Hello @ekdahl, thanks for submiting the issue. I believe it's duplicate to #7. 
I will work on it and try to release a version until next week since it's an improvement that seems to be important to many.

Rickedb, 2022-06-06:
Enhancement made at version `5.0.0`, however when packing and revision is set to `0`, it will pack as empty.

Since 0 or empty are the same, we had to choose one between them, since it would make more complex to specify revision as nullable int (`int?`). 

Why? Because it would worth nothing to have this feature, nonetheless, it was going to make our lives harder checking if revision has value or not to access `.Value` property.

For more information, please check release notes: https://github.com/Rickedb/OpenProtocolInterpreter/releases/tag/5.0.0

closed 2022-06-06 by Rickedb