Rickedb/OpenProtocolInterpreter #45: Description of MIDs

ekdahl, 2021-02-10:
I'm missing a description property or dictionary with descriptions for the different MIDs. Something like the header from the specification, so for MID 0061 it would be "Last tightening result data". It's useful for logging what was sent or received.

Rickedb, 2021-02-25:
Hello @ekdahl, sorry for the delay.
It's a good idea for an extension project, however it's not a "core thing" since it's only useful for logging.
It could be something like `OpenProtocolInterpreter.Logging`, such as Serilog `Serilog.Sinks`. But I don't see it as part of the core library.

Feel free to share your ideas

ekdahl, 2021-02-26:
Hi, ok I see.
I did a MID interpreter that outputs the properties for easier viewing and was missing a description of what the MID is for (the application doesn't need to know anything about what MIDs can be parsed). Anyway for this particular use, I used a dictionary with descriptions instead which works fine but all the descriptions needed to be in my application instead of in the library where I think it would be a useful addition.

		public static Dictionary<int, string> MidDescriptions { get; set; } = new Dictionary<int, string>
		{
			{Mid0001.MID, "Application Communication start"},
			{Mid0002.MID, "Application Communication start acknowledge"},
			...

IngeRuppio, 2021-08-11:
I did the same, but only for the Mids, supported by my software. So yes, I would like to see this included as well.
Because people may add their own custom Mids (as I did), there should be an easy way to add these to the one global dictionary (or whatever).