Rickedb/OpenProtocolInterpreter #51: Bugs processing MID0106

Raykler, 2021-05-06:
With the newer versions the MID0106 is not working properly, here an example (replace _ with space):
`03370106____________01030201030000723323040105SU_ST2.1_PwMACS_____062021-05-06:13:46:14075008Mode_50_____________09110011________________________________________1202130114115_16117_______18-2885.719_______20_______21_______22_______130214115_16117_______18-2885.819_______20_______21_______22_______2301Data_No_Station_____I_100000002438`

This is an example of a loosen, the BoltsData, in the angle, should return -2558.7
But depending on the version returns:
V: 3.4.0 => angle 0
V: 3.3.0 => angle 0
V: 3.2.2 => angle -28857 (no decimal, probably due to CultureInfo)

Rickedb, 2021-05-07:
All numbers `< 0` are getting parsed to 0. Going to fix it for the next version.

> Btw, what a huge angle value 😂

Rickedb, 2021-05-07:
Occurs only when parsing, packing method does not fail to generate with signal

Rickedb, 2021-05-07:
Fixed at version `3.4.1`. Published right now at NuGet, probably be available in a few minutes/hours

closed 2021-05-07 by Rickedb