Rickedb/OpenProtocolInterpreter #53: Loop may crash in class ParameterSetIdListConverter

IngeRuppio, 2021-05-17:
OpenProtocolInterpreter.Converters, class ParameterSetIdListConverter, public override IEnumerable<int> Convert(string value):
for (int i = 0; i < value.Length; i += 3)
must be changed into 
for (int i = 0; i < value.Length - 2; i += 3)
Otherwise the method crashes if value has the max. length (9977 characters or so).

Rickedb, 2021-05-17:
I've tried to test this case, but couldn't simulate it, generating a package with 999 ParameterSets _(which is the max parameter sets)_, but it seems to parse correctly.

Do you have any example of it?

IngeRuppio, 2021-05-19:
Caused by a bug in a test program, I received a string with 9977 characters, all \0. 
Therefore, at the end of the loop, value.Substring(i, 3) tried to access data beyond the end of the string.
I think, the problem occurs, if value.Length is not a multiple of 3 (it should be, but if not: booom).

Rickedb, 2021-05-31:
Well, it's not possible to have more than 999 parameter sets, so it wouldn't reach this length.
Also, if there is a problem with the package, I don't think it's bad for the library to throw the exception up. What we might consider is to throw a custom exception instead of a generic `IndexOutOfRangeException`

Rickedb, 2021-09-14:
Closed due to inactivity. If necessary, reopen it.

closed 2021-09-14 by Rickedb