Rickedb/OpenProtocolInterpreter #56: Mid0061 TighteningErrorStatus - wrong bit field encoding?

IngeRuppio, 2021-05-17:
If I set the "Rehit" flag in the TighteningErrorStatus entry of Mid0061, I get an encoded value of 4194304.
Rehit is documented to be bit 18. Because they seem to count from 1, it should be 2^17 = 131072. But it isn't. Why?
And the bits in StrategyOptions count from 0 in the documentation, but you start with 1. Hmmm...
As long as encoding and decoding both is done by your OpenProtocolInterpreter, this may work. But what happens, if another code counts and converts in a different way. The bits all would be messed up and nothing would work.

Rickedb, 2021-05-18:
`SetByte` method from `BitConverter` is setting bits "left to right" orientation, not "right to left", which causes the incorrect byte array.

IngeRuppio, 2021-05-19:
So this is a bug on your side? 
I cannot verify this, because both applications, I use for testing, do not support bit fields.
And I cannot find any details in the OpenProtocol documentation.
Do you have any detailed documentation on the different bit fields?
A potential client is waiting for a proof of concept demonstration...

Rickedb, 2021-05-19:
Yes, it is, just commenting for documenting purpose. Also tagged as a bug.
I'm going to fix it ASAP, probably until the weekend!

IngeRuppio, 2021-05-19:
Hello Henrique,

I’m afraid, there is a major problem with encoding and decoding of bit fields as they are used in Mid0061 and others.

You seem to encode theses bit fields binary, which results in non ASCII characters including 0x00.
Only the Rehit bit set results in (hex values): 32 30 00 00 40 00 00 00 00 00 00 00
I think, this is wrong. Bit fields should be encoded as integer values to ASCII characters.

Neither the binary nor the ASCII method is documented in the Open Protocol manual.
So I rely on some hints, I found:

  1.  MID 0033, Revision 3, Job list:

Job step type = Two ASCII characters, range 00-99     Batch step = 1     Reserved = 2-6

Including the reserved, there can be up to 6 bits, which creates a maximum value of 63 – 2 ASCII characters.

The examples shows, that Bit 1 (the lowest bit) is encoded as “01” (lowest bit on the right).

  1.  MID 0061 and 0065, Revision 6, Tightening error status 2:

Including the reserved, there are up to 32 bits documented. 32 bits give a maximum decimal value of 4294967295.

This requires 10 ASCII characters, exactly the size, which is specified for this field (517-526).

Just a coincidence? I don’t’ think so. A binary encoding only requires 4 bytes of the available 10. Makes no sense for me.

  1.  Chapter 2.2 says: “Most of the information sent over the communication links is in ASCII format. Some messages can also contain binary data and this is described at each MID.”

But there is no description of non ASCII in the Mids 0061, …

From my understanding, the Rehit bit in Mid 0061 (Bit 18, counting from 1) must be encoded to 2^17 = 131072.
The encoded entry therefore should be “200000131072” and in hex: 32 30 30 30 30 30 31 33 31 30 37 32

I`m not a 100% sure, but let’s say 95%, no 98%. 😉

Also there seems to be a different result by Pack() (decimal character digits, but wrong number) and PackBytes() (binary)

Please carefully think about it. It’s not enough, if you package can send and receive correctly.
The communication must work with other programs from other developers and companies as well.

Bye,

Inge

Rickedb, 2021-05-30:
For `Mid0061`, I'm sure it's been converted to integer before we convert to an ASCII value, that ensures we are not encoding wrongly.

What happens is that, it parses correctly, however when packing, it's creating the byte array reversed, for example, in StrategyOptions we have a Unit tests that it's integer value is `67` which stands for Torque, Angle and Rundown. _(Bytes 0, 1 and 6)_

When packing the integer value was `194`:
67 => 0100 0011
194 => 1100 0010

This is a defect in `BitConverter` class, that threats it reversed. I'm already working for the solution.

This affects `StrategyOptions`, `TighteningStatus` and `TighteningStatus2`.

IngeRuppio, 2021-05-31:
Pack() now seems to create a correct entry for TighteningErrorStatus (Rehit now is "0000131072"). 👍 
But PackBytes() still creates a binary entry, so the Integrator cannot read Mid0061 correctly. 👎

IngeRuppio, 2021-06-01:
In the Controller, I now use
        byte[] msg = Encoding.ASCII.GetBytes(mid.Pack());
instead of
        byte[] msg = mid.PackBytes();
and it works. The Controller can send Rehit and the Integrator does receive it correctly.

But I have no idea, why you use totally different code for Pack() and PackBytes() - both should always create identical results (besides string vs byte array). Just delete the "public override byte[] PackBytes() {}" and it should work fine!? Same for Parse(). Just an idea, only you know...

Rickedb, 2021-06-01:
I'm already working on it.,The reason of it is what you said _13 days ago_, besides the use of a bit approach, it is always sent as an ASCII.

What happens is that when using `PackBytes()` it just creates the byte array in it's "pure form".
What it needs to do is to create the byte array, convert to an Int32 or Int64 value and then encode the generated number (padded left with 0) to ASCII again.

`Mid0061` is already fixed, however I'm working on the same issue for `Mid0065` now

closed 2021-06-01 by Rickedb