Rickedb/OpenProtocolInterpreter #57: LEFT_PADDED missing in Mid0002

IngeRuppio, 2021-05-17:
A while ago, you added LEFT_PADDED to CELL_ID and CHANNEL_ID - fine, these were missing in earlier versions.
But SYSTEM_TYPE and SYSTEM_SUBTYPE do need LEFT_PADDED as well - please add these!

Rickedb, 2021-05-18:
Couldn't achieve the same behaviour here. Both seems to be left padded in the current version:

```csharp
...
{
    5, new  List<DataField>()
            {
                new DataField((int)DataFields.SYSTEM_TYPE, 163, 3, '0'),
                new DataField((int)DataFields.SYSTEM_SUBTYPE, 168, 3, '0')
            }
},
```

IngeRuppio, 2021-05-19:
I misused these fields to transfer some private identifiers. Depending on the number of digits of the values, they got messed up.

closed 2022-03-20 by Rickedb