Rickedb/OpenProtocolInterpreter #59: Mid0011 does not work because of incorrect length calculation

IngeRuppio, 2021-05-17:
in Parse() you set the size:

GetField(1, (int)DataFields.EACH_PARAMETER_SET).Size = package.Length - GetField(1, (int)DataFields.EACH_PARAMETER_SET).Index;

This is wrong, because "package" includes the Message End termination NUL.
If you add a "- 1", Mid0011 works fine.
My fix is a bit more secure:

int length = package.Length - GetField(1, (int)DataFields.EACH_PARAMETER_SET).Index;
if (package.Length > 0 && package[package.Length - 1] == '\0')
{
  length--;
}
GetField(1, (int)DataFields.EACH_PARAMETER_SET).Size = length;

Rickedb, 2021-05-18:
Since the header length property excludes NUL termination, it's more reliable and performance will be better _(just a little)_

Documentation about header:
|Message part|Byte|Parameter|Value|
|---------------|-----|-----------|------|
|Header          |1-4  |Length     | **The length is the length of the header plus the data field excluding the NUL termination.**<br>The header always includes information about the length of the message. The length is represented by four ASCII digits (‘0’…’9’) specifying a range of 0000 to 9999. <br> When using the message linking functionality the length represents the length of each message part number.<br>When having one ASCII part followed by an binary part the length is the total length of the message.|

IngeRuppio, 2021-05-19:
You are right for the header field "Length", but package.Length does not access the header field Length.
package ist the string, which contains the entire message and this includes the NUL termination. Therefore your calculation is wrong and Mid0011 does not work.

closed 2021-05-31 by Rickedb