Rickedb/OpenProtocolInterpreter #63: Request for Mid0701 (Tool list upload reply)

IngeRuppio, 2021-05-19:
I would like to see Mid0701 (Tool list upload reply) supported by your package.
It is used to upload the Tool list according to Mids 10-13 for Psets. 
Okay, it's part of OpenProtocol v2, but it would be very helpful for me (and others).
I tried to implement it as custom Mid by my own, but because of the repeating items it is somewhat complicated for a girl not familiar with you package internals. 
For you, it should be easy, bacause you already implemented similar lists in the Mids 106/107.

Rickedb, 2021-05-23:
I will map new Mids from OpenProtocol V2, so I can add it to the roadmap and release a new version.
If possible, please provide an example of the `Mid0701`, that helps when generating UnitTests.

Rickedb, 2021-09-06:
@IngeRuppio  Got any ASCII examples ?

IngeRuppio, 2021-09-06:
[MID0006.cs.txt](https://github.com/Rickedb/OpenProtocolInterpreter/files/7113616/MID0006.cs.txt)
This is my code, how I respond to a Mid0701 request via Mid0006.

closed 2021-09-09 by Rickedb