Rickedb/OpenProtocolInterpreter #77: Issue MID 0245

tecnologic-srl-roberto-guerzoni, 2021-09-23:
I'm using your excellent library to realize an Open Protocol Simulator. During my development i noticed an issue into MID 0245. I suppose the issue derive from a wrong cut and paste.
In my opinion the code should be fixed into this point
    public class Mid0245 : Mid, IPLCUserData, IIntegrator
    {
        private readonly IValueConverter<int> _intConverter;
        private const int LAST_REVISION = 1;
        public const int MID = 245;

        public int Offset
        {
            //old:get => GetField(1, (int)DataFields.USER_DATA).GetValue(_intConverter.Convert);
            //old:set => GetField(1, (int)DataFields.USER_DATA).SetValue(_intConverter.Convert, value);
			get => GetField(1, (int)DataFields.OFFSET).GetValue(_intConverter.Convert);
			set => GetField(1, (int)DataFields.OFFSET).SetValue(_intConverter.Convert, value);
        }

I'm not sure the USER_DATA field must be filled to 200 chars.
The documentation say:
<html>
<head><style>

<body>

Offset | 21-23 | Three ASCII digits Range 000-099Specify the address offset in number of bytes for theuser data in the PLC.The data is written to address 13000 + Offset in thePLC
-- | -- | --
User data | 24-max 223 | Minimum 2 and maximum 200 ASCII characters.See MID 0240 for a description.The maximum length for the field is 200 – 2 * Offset

 
<br style=" font-style: normal; font-variant: normal; font-weight: normal; letter-spacing: normal; line-height: normal; orphans: 2; text-align: -webkit-auto; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; ">

</body>
</html>
Can you confirm the need of that padding?

Have a nice day

Roberto Guerzoni

Rickedb, 2021-09-23:
Hello Roberto, good to hear that the library is helping you.
Answering your questions:

1. Offset field does have a bug and it was probably a copy + paste issue 😂
2. About User data field: no, it's not obligatory to fill with 200 chars. In addition, the library (still) does not watch over the minimun and maximun values.

closed 2021-09-23 by Rickedb

tecnologic-srl-roberto-guerzoni, 2021-09-24:
Hello Henrique
                thank you for your clarification

Have a nice day
Roberto Guerzoni