Rickedb/OpenProtocolInterpreter #79: Mid0076 (Alarm status) always creates revision 3

IngeRuppio, 2021-10-06:
If a Mid0076 is created using a revision 1/2/3 constructor, it always is revision 3, because there are no revision arguments in the constructors. Therefore always LAST_REVISION, which is 3, is used.

By the way, revision 3 cannot be used, because Mid0070 (Alarm subscribe) only supports revision 2.

IngeRuppio, 2021-10-06:
// Example:
// this creates an illegal command (should be rev1, but is rev3 with missing ToolHealth data):
            mid = new Mid0076(alarmStatus, errorString, controllerReady, toolReady, alarmTime);

// this works
            mid = new Mid0076(1)
            {
                AlarmStatus = alarmStatus,
                ErrorCode = errorString,
                ControllerReadyStatus = controllerReady,
                ToolReadyStatus = toolReady,
                Time = alarmTime
            };

IngeRuppio, 2021-10-06:
And if you add revision 3 to Mid0070, Mid0071 has to be updated with revision 3 as well.

Rickedb, 2021-10-06:
Hello @IngeRuppio, thanks again for providing another issue.

Since there is a way to make it work, I will be fixing this issue in the [next project](https://github.com/Rickedb/OpenProtocolInterpreter/projects/7) which is to implement new updates from `SR 2.14` and add some Revisions that were forgotten in earlier updates.

Otherwise, if it's causing a big impact into applications, I might fix it before the project.

Is it a huge problem for you to wait until the next project update is done?

IngeRuppio, 2021-10-07:
It is a big issue here, because two large companies are affected and the workflow is blocked.
But because I`ve been able to use the workaround from my comment above, you have all the time you need to fix it.

Rickedb, 2021-10-15:
Hello Inge,

Had to fix another issues that had no workaround and already implemented the necessary fix to Mid0076. Version `3.6.2` available!