Rickedb/OpenProtocolInterpreter #86: Mid0061 timestamp get 0001-01-01T00:00:00

bill024, 2022-03-17:
"TorqueControllerName":"E17051","GunTimeStamp":"2022-03-14T15:36:39","ProductID":"H2221283/A","TighteningCount":9,"Torque":706.0
"TorqueControllerName":"E17051","GunTimeStamp":"2022-03-14T15:36:51","ProductID":"H2221283/A","TighteningCount":10,"Torque":705.0
"TorqueControllerName":"E17051","GunTimeStamp":"2022-03-14T15:37:12","ProductID":"H2221283/A","TighteningCount":11,"Torque":708.0
"TorqueControllerName":"E17051","GunTimeStamp":"2022-03-14T15:37:17","ProductID":"H2221283/A","TighteningCount":11,"Torque":696.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":12,"Torque":701.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":13,"Torque":715.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":14,"Torque":697.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":15,"Torque":698.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":16,"Torque":697.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":17,"Torque":701.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":17,"Torque":701.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":18,"Torque":705.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":19,"Torque":698.0
"TorqueControllerName":"E17051","GunTimeStamp":"0001-01-01T00:00:00","ProductID":"H2221283/A","TighteningCount":20,"Torque":697.0
"TorqueControllerName":"E17051","GunTimeStamp":"2022-03-14T16:00:00","ProductID":"H2221283/A","TighteningCount":21,"Torque":719.0
"TorqueControllerName":"E17051","GunTimeStamp":"2022-03-14T16:00:02","ProductID":"H2221283/A","TighteningCount":21,"Torque":701.0
"TorqueControllerName":"E17051","GunTimeStamp":"2022-03-14T16:00:11","ProductID":"H2221283/A","TighteningCount":22,"Torque":700.0
"TorqueControllerName":"E17051","GunTimeStamp":"2022-03-14T16:00:17","ProductID":"H2221283/A","TighteningCount":23,"Torque":696.0

As shown in the data,when the Software receive torque gun data,the parameter TimeStamp sometimes receive “0000-01-01T00:00:00”,please help analyze.

The OpenProtocol instructions is MID 0061（Last tightening result data）

Rickedb, 2022-03-17:
Hello @bill024, do you have any logs which can help us to analyze?

Rickedb, 2022-06-06:
Closed due to inactivity

closed 2022-06-06 by Rickedb