Rickedb/OpenProtocolInterpreter #87: SimpleTCP.Net5

danieldorneles1980, 2022-03-18:
Boa tarde. 
Gostaria de parabenizar pelo trabalho com este componente. 
Nos próximos meses pretendo implementar a MID0900 e enviar caso queiram publicar.

Ontem alterei o pacote SimpleTCP para a versão SimpleTCP.Net5 (1.0.25), e usei o seguinte construtor:
SimpleTcpClient client = new SimpleTcpClient { StringEncoder = System.Text.Encoding.ASCII, Delimiter = 0x00 } ;

Porém, pelo que percebi tenho quase todos (senão todos) os pacotes duplicados. 
Já chegaram a utilizar a versão que mencionei? Tem idéia do que pode estar causando este problema?
Inicialmente pensei que pudesse ser problema com ReadLoopIntervalMS, mas depois verifiquei que ele estava com valor 10 como na versão que utilizam.

Rickedb, 2022-03-20:
Boa noite @danieldorneles1980,

Fique a vontade para fazer um pull request em relação a implementação do `Mid0900`.
Não cheguei a fazer a utilização dessa versão do SimpleTCP. Mas é bem provável que tenha a ver com o pacote do próprio SimpleTCP.

Recomendo você verificar se não está por algum acaso duplicando o bind do evento `DataReceived`.
E sempre que der um `Dispose()` na sua classe garanta em dar o unbind (`-=`) no evento também, pois o GC não faz isso automaticamente.

Rickedb, 2022-06-06:
Closed due to inactivity

closed 2022-06-06 by Rickedb