Rickedb/OpenProtocolInterpreter #88: Missing Link Level acknowledging or not clear the usage example

roniinterconetica, 2022-04-12:
First I would like top congratulate you for this project. It's very nice!
Based on the vresion 2.8 and newers versions of protocol , (3.2.2 Link Level acknowledging method from OP spec. 2.0 page 38)
Is possible to use sequence in headers to check correctly the message interchange. 
As the protocol says 
_**"OBS! It is really recommended and almost a mandatory to use this method instead of Application Method acknowledging due to a lot of customer reported problems during the years of OP existence, which will be solved by this OBS!."**_

For Example :  

[image]

How  should I use the OpenProtocolInterpreter with sequence acknowledging ? 
Is it possible and implemented in all messages ?

Thank you a lot !

Rickedb, 2022-04-17:
Hello @roniinterconetica and sorry for the delay, been out since last week.

The Header part was developed at 2017 and I probably misunderstood the concept, which was translated wrongly as `UsedAs` in the library. 

But answering your question, yes, it's possible to use, however it will be annoying for you to use it in the way it's working right now.

I'm going to work in a new version to fix this.

roniinterconetica, 2022-04-19:
You are great @Rickedb !

Rickedb, 2022-06-06:
@roniinterconetica  Sorry for the delay, was in a rush since April. But version `5.0.0` is out now, any problems please contact me or write an issue.

For more information, please check the release notes: https://github.com/Rickedb/OpenProtocolInterpreter/releases/tag/5.0.0

closed 2022-06-06 by Rickedb