Rickedb/OpenProtocolInterpreter #90: MID1201

sosborne-control2k, 2022-07-05:
Hi, I get Object Reference not set to instance error on 1201 but not 1202 - adding:

_ProcessDataFields(package);_ 

on line 113 fixed for me.

Thanks for great package.

Rickedb, 2022-08-08:
Hello @sosborne-control2k  Thanks! Gonna try to update it as soon as possible!

OscardLucas, 2022-11-14:
Additional this bug i found two more in the MID1201.
PD. ProcessDataFields method should be added on line 125, after set the Index and size of Fields that are assigned at run time

**Description:**
On Parse Method for MID1201
VariableDataFields throws null exeption when trying to convert dataFieldsListField Value to List. This parameter is not sent when number of data fields is 000, that causes null exception.
As documentation especifies, section 5.9.1.4 on Table 120, Data fields section: _"This section is repeated Number of data fields times. If Number of data fields = 000, this section is not sent"_

**Solution**
I evaluate the NumberOfDataFields before assign the value.
Replace
`VariableDataFields = _varDataFieldListConverter.Convert(dataFieldListField.Value).ToList();`
with
`VariableDataFields = NumberOfDataFields > 0 ? _varDataFieldListConverter.Convert(dataFieldListField.Value).ToList() : VariableDataFields;`

 -------------------------

**Description**
TIME DataField for MID1201 is not properly parsed due incorrect size property.

**Solution**
On RegisterDataFields method for MID1201 change TIME Field size.
Replace
`new DataField((int)DataFields.TIME, 36, 4, '0', DataField.PaddingOrientations.LEFT_PADDED, false),`
with
`new DataField((int)DataFields.TIME, 36, 19, '0', DataField.PaddingOrientations.LEFT_PADDED, false),`

**Additional Information**
Working with MTFocus 6000 controller
With this response both errors can be reproduced.
"006912010010    00  00200100000001532022-11-14:09:35:1000000100001000\0"

closed 2023-05-31 by Rickedb