Rickedb/OpenProtocolInterpreter #92: Not setting some properties produces invalid output

ekdahl, 2022-09-05:
When not setting some properties the output generated when packing to a string is invalid. The length in the header says what it should be, but the produced output is missing characters from the unset properties.
For strings for example, wouldn't it be better to output a string of the correct length filled with spaces if the property has not been set?
Integers could be 0, booleans could be false etc.
I understand that there could be situations where it is hard to define a default value, but it would be good to know that a valid output is produced, without setting all properties.

Also, when setting string that is longer than max length, the length is not limited to the specified length.

Sample console application (but applies to many MIDs):
```
using OpenProtocolInterpreter;
using OpenProtocolInterpreter.Communication;

var midInterpreter = new MidInterpreter().UseAllMessages();

Console.WriteLine("'" + new Mid0002(1).Pack() + "'");
// Invalid output: '00570002001         010203'
// Message too short

var mid0002 = new Mid0002(1) { CellId = 0, ChannelId = 0, ControllerName = String.Empty };
Console.WriteLine("'" + mid0002.Pack() + "'");
// Valid output: '00570002001         010000020003                         '

mid0002 = new Mid0002(1) { CellId = 0, ChannelId = 0, ControllerName = "String that is longer than 25 characters" };
Console.WriteLine("'" + mid0002.Pack() + "'");
// Invalid output: '00720002001         010000020003String that is longer than 25 characters'
// String not limited to 25 characters, length in header not according to specification
```

Rickedb, 2022-09-05:
The default char for padding is always `' '`, variable probably is not initialized at data fields.

Going to check it, thank you for submiting it!

ekdahl, 2025-01-29:
I guess the initialization can be done as a general fix in the [DataField](https://github.com/Rickedb/OpenProtocolInterpreter/blob/master/src/OpenProtocolInterpreter/DataField.cs) class, if that ensures that there is always a value assigned (e.g. filled with spaces for strings).
Where should this preferably be done, in the constructor or GetValue?