Rickedb/OpenProtocolInterpreter #93: Add pretty printing

melvyniandrag, 2022-10-19:
Does this library have any way to create pretty strings from a message? Would be nice to say 
```
var msg = Parse(string)
Console.WriteLine(msg.asPrettyString())
```
or something like this.

Would you like to have this in the library?

Rickedb, 2022-10-21:
Hi @melvyniandrag, you mean printing MID data such as a json output or something like mentioned in #45?

melvyniandrag, 2022-10-25:
Hi @Rickedb yes exactly that. Any format really, json, csv, anything.

For now I've got stuff like this:
```
           String ret = "";
            ret += "MID0001 - Start Communication" + "\n";
            ret += "Length: " + mid.Header.Length + "\n";
            ret += "Mid: " + mid.Header.Mid +"\n";
            ret += "Revision: " + mid.Header.Revision +"\n";
            ret += "No Ack Flag: " + mid.Header.NoAckFlag +"\n";
            ret += "StationID: " + mid.Header.StationId +"\n";
            ret += "SpindleID: " + mid.Header.SpindleId + "\n";
            ret += "Spare: " + "Not implemented in library" + "\n";
            return ret;
```
Which is time consuming to write because I'm handling many MIDs.

I imagine the library has a clean way to loop over the values and stick them in a nice, readable format.

I only looked briefly, but I think there is a List<DataField> in the classes that could be looped over to create a json, csv or whatever. Seems like it would be easy to implement this in one small function.

Rickedb, 2022-10-26:
I understand that logging is extremely necessary when dealing with OpenProtocol. 
Although it's not the main purpose of the library, I still want to develop some easy way to print/log OP data to help out.

These months _(or this year completely)_ are being a little tough to me due to many side projects, which is making me a little bit away of OpenProtocolInterpreter project. But I promise this and other features will come as soon as possible.