Rickedb/OpenProtocolInterpreter #95: How to use multispindle mid

anandanatarajan, 2022-11-17:
Mid0100 for multispindle result subscribe and multispindle result, acknowledge and unsubscribe are all showing as internal class in version 5.0, how to use any examples, 
Thanks for the support

Rickedb, 2022-11-17:
Hi @anandanatarajan, what do you mean by showing as internal? As far as I've checked, they're all public classes.

anandanatarajan, 2022-11-22:
hi @Rickedb , sorry that was mistake on my side, mixed up with multiple version of openprotocolinterpreter. 
thanks for taking time to respond. on side note, can you share any example on how to get data from multispindle devices like powermacs4000, the PowerMacs MiD105 is returning 'TIMEOUT' error, any help?

anandanatarajan, 2022-11-24:
hi @Rickedb ,

Thanks for taking time to help, actually the controller only responds to 90 and 91 mid which is multispindle status , i am unsure about how to proceed further, i.e. how to get torque value after getting status , since i am working with these machines for the first time. Thanks once again.

Rickedb, 2022-11-28:
Here in our project we're using MID0105 subscription to obtain Mid 0106. However we select job with MID 240, which, if I'm not mistaken, the PMs were not handling MID 0030 and the solution made by the Atlas' developer was to use MID 0240.

Don't you have any Atlas support to help you out? PMs are way different from PF, seems that they can program their behaviour more "freely"

anandanatarajan, 2022-12-03:
Hi @Rickedb ,

Thanks for taking time to answer questions, yes we have asked Atlas guys for support. PM series is a kind of bottleneck for us, Anyway i greatly appreciate the efforts you have put on making our lives easier. Thanks very much.
we can close this thread as of now.

closed 2022-12-03 by anandanatarajan