Rickedb/OpenProtocolInterpreter #96: MIDs Revisions

asidadon, 2022-12-31:
**No issue, More like a question**
Hi Enrique,
first, thanks for this amazing interpreter, it is very helpful, i wish to ask a question please:
i am currently working with Revision 5 of the interpreter and I'm trying to use MID 0061 on my massage, which is revision 1.
from looking at the interpreter code it looks like it always uses the massage as if it was Revision 8, is that correct? if it is, can you help me understand how can i properly read my Revision 1 message? 
thanks, Asi.

Rickedb, 2023-01-01:
Hello @asidadon, 

When creating/sending a mid , by default the library will always assume you're using the highest revision if you're using default constructor. So, if you want to specify another revision you should specify it at the constructor. e.g.: `new Mid0061(1)`.

When parsing it will considers the revision sent by controller/integrator.

However, something tells me this wont answer your doubt. 
When you say "properly read", you mean which properties to use ?

asidadon, 2023-01-01:
hi @Rickedb,
thanks for the quick reply, actually you have answered my question.
my tightening result messages are revision 1 and in the documentation it looks like revision 1 is different from 2 and on, so i thought it might cause wrong parsing of the message string.
anyway i have successfully applied your explanation and it worked good, thanks.
Best Regards

Rickedb, 2023-01-01:
Just clearing things: Yes, revisions are different, usually they have additional information as far as revisions are increasing.

Take sometime to download read the PDF documentation [here](https://github.com/Rickedb/OpenProtocolInterpreter/blob/master/docs/OpenProtocol_Specification.pdf) it will explain to you how it works in dept.

closed 2023-01-06 by Rickedb