Rickedb/OpenProtocolInterpreter #103: mid60

Mr4g, 2023-06-07:
Unfortunately mid 0060 its class is not completed does anyone have such a class completed?

Rickedb, 2023-06-08:
By default in version 6.0.0 all Mids are set to revision 1 when using the empty constructor. 
However set it by the another overload of the constructor `Mid0060(int revision, bool noAckFlag = false)`

Mr4g, 2023-06-08:
[image]

doesn't work

Rickedb, 2023-06-08:
Seems that you're not familiar with C#. 

Try: `new Mid0060(6, false)` or another possible option is: `new Mid0060(revision: 6, noAckFlag: false)`

Mr4g, 2023-06-08:
you're right I'm just learning, but before that I also tried like you, but I forgot to pay attention to what parameter the constructor takes and noAckflag is of type int 

[image]

Rickedb, 2023-06-09:
You're probably using an older version of the library which used to be an `int?`, versions 5.X.X or greater is boolean. 
But it's still the same approach.

closed 2023-07-03 by Rickedb