Rickedb/OpenProtocolInterpreter #105: MID0038 - Select Job Invalid Data

URSIRob, 2023-08-24:
I wanted to test the Library You created as it could help me a lot in project that I'm helping to my friend to work with the Ingersoll Rand  Controller and in general all is working great except I can't select the job, I'm receiving message Invalid Data all the time. I checked the message that is being send and it is "00240038009" for JOB ID 9 bu when I check the Opent Protocol specification there is mentioned that 0024 mean that this is a Revision 2 so from the documentation it should be maybe 4 digits for the job ID  and maybe the proper message should be "002400380009"? Maybe this is somehow controlled by some option or it is a matter of the Controller that I configured wrong but can you take a look and help me with figuring this out?

Thank you in advance.

[image]

Rickedb, 2023-08-25:
Hello @URSIRob, thanks for submitting it.
The problem is with the code, since the breaking changes of the version 6.0.0, all constructors that set datafield were removed.

In `SendJobCommand.cs`:
https://github.com/Rickedb/OpenProtocolInterpreter/blob/d082e88a94a23124cc73d3c83a06f4a9ef3caf02/sample/OpenProtocolInterpreter.Sample/Driver/Commands/SendJobCommand.cs#L16-L29

The `new Mid0038(jobId)` is calling the `Mid0038(int revision)` constructor and setting the revision as 9, not the Job Id and that's why it's messing up.

closed 2023-08-25 by Rickedb

URSIRob, 2023-08-25:
Thank you. After your fix It's working great, thanks to your comment I understood it better. Great job, one more time thank you!

commit b48b49d4 (2023-08-25): Fixes #105