Rickedb/OpenProtocolInterpreter #106: Questions on Protocol and Specification Versioning

dekenless, 2023-10-02:
Below is the description from the OP documentation no how they do version, release & revision specification -- when using the OpenProtocolInterpreter it appears that one only really needs to be concerned with the revision when dealing with specific MIDs (i.e. the version and release are more or less irrelevant).  Is that correct?

I regards to OP 2, that largely seems to be additional MIDs and dealing with variable message data size--is OP 2 supported?

[image]

Rickedb, 2023-10-08:
Hi @dekenless, you're right, the OpenProtocolInterpreter version has **no relation** with OP versioning.
Since we might face bugs that has no relation with the protocol itself (an internal bug for example), it would be difficult to keep versions equal and would cause more confusion. *(If you have a suggestion on how to handle this, I would appreciate it)*

As far as I know, most of MIDs are compatible with previous versions and the lib is not handling these compatibility issues yet.

The reference of OP version for the latest release of OpenProtocolInterpreter is 2.14.0 ([see docs here](https://github.com/Rickedb/OpenProtocolInterpreter/blob/master/docs/OpenProtocol_Specification.pdf)).
If there is another version published, I would ask you to provide the docs, so I can develop the new modifications or new MIDs at the library. Optionally, you can do it by yourself and make a PR too.

Check [unavailable mids section](https://github.com/Rickedb/OpenProtocolInterpreter#list-of-still-unavailable-mids) for further information on which are not implemented *(most of them due to lack of example for testing)*.

dekenless, 2023-10-09:
Thanks for the reply.

Attached is the latest version that I know of, 2.17 I believe.
[OpenProtocol_Specification_R_2.17.0.pdf](https://github.com/Rickedb/OpenProtocolInterpreter/files/12847933/OpenProtocol_Specification_R_2.17.0.pdf)