Rickedb/OpenProtocolInterpreter #112: reference assemblies for .NETFramework,Version=v4.8 were not found

Sourab1801, 2024-04-25:
PS E:\sourabh_data\Project\openProtocolInterpreter\OpenProtocolInterpreter\sample\OpenProtocolInterpreter.Sample> dotnet run 

C:\Program Files\dotnet\sdk\8.0.204\Microsoft.Common.CurrentVersion.targets(1248,5): error MSB3644: The reference assemblies for .NETFramework,Version=v4.8 were not found. To resolve this, install the Developer Pack (SDK/Targeting Pack) for this framework version or retarget your application. You can download .NET Framework Developer Packs at https://aka.ms/msbuild/developerpacks [E:\sourabh_data\Project\openProtocolInterpreter\OpenProtocolInterpreter\sample\OpenProtocolInterpreter.Sample\OpenProtocolInterpreter.Sample.csproj]

The build failed. Fix the build errors and run again.

Rickedb, 2024-04-25:
Hello @Sourab1801, since the sample is a .Net Framework 4.8 Window Forms application, you cannot run it via `dotnet run`. 

As far as I know, you cannot run the application at VSCode, you'll need a Visual Studio Community or Rider at least.

Rickedb, 2024-05-30:
Closing due to inactivity, please reopen if necessary or any updates.

closed 2024-05-30 by Rickedb