Rickedb/OpenProtocolInterpreter #115: MID 0060 Last tightening result data subscribe; It works but it doesn't show torque or angle.

Leonel7, 2024-07-23:
I can get all the info for my last tightening result but the torque and angle is always on zero.
But I dont know if is a configuration problem on mi powerfocus 6000 because in the report window it doesn't show me the torque or the angle either.
[image]
[image]
[image]

Rickedb, 2024-07-26:
Hi @Leonel7, sorry for the delay in response.
Since the package sent is all zeros, this is probably an issue with the controller, not with the package at all. It could be the firmware as well.

When applying the torque, does it shows the correct values at least?

Leonel7, 2024-07-26:
> Hi @Leonel7, sorry for the delay in response. Since the package sent is all zeros, this is probably an issue with the controller, not with the package at all. It could be the firmware as well.
> 
> When applying the torque, does it shows the correct values at least?

Don't worry about the delay.
Yes, it shows the correct value in the live result and the value changes every time I torque. But I have tested it with all revisions and it is always 0 in MID 0061.
The people from atlas tell me that the problem is because of my license. I have an station control license, they told me that my license dosent record results and thats the problem and they want to sell me a critical control license.
Okey, my license doesnt record results, so maybe mesages like "Old tightening results (MID 0064)" wont work. But if the torque and angle are right there in the live result there should be an MID to take that result... I'll get support from the Atlas guys in two hours to see if there's any solution, because the Critical Control license is not compatible with my tools.

I will update the solution in case anyone else has the same problem.

thanks!!

Rickedb, 2024-07-27:
Makes sense to be, although I've never seen such a behaviour even on the last time I worked with PF6K.

But if so, please update here. It will help other people that might run over the same issue.

Leonel7, 2024-07-29:
> Makes sense to be, although I've never seen such a behaviour even on the last time I worked with PF6K.
> 
> But if so, please update here. It will help other people that might run over the same issue.

Well, the guys from Atlas told me that with the "station control" license, I can't even send the torque/angle value to the PLC with a fieldbus card because it is "protected", even if I use some Atlas software (for example sqs) I wouldn't get that data.
Solution provided by Atlas: buy 6 "critical control" licenses for my 6 controllers = $18,000 USD.
I'm going to check out MID0008 and MID1201/1202. They seem to be useful for getting torque and angle results as well, but they don't seem to be compatible with PF6k.
Thank anyway!!!

Rickedb, 2024-07-29:
This is horrible. You must buy the equipment and pay for license as well?
At least these licenses are one shot? Or do you have to pay for it like monthly/annually?

Leonel7, 2024-09-22:
> This is horrible. You must buy the equipment and pay for license as well? At least these licenses are one shot? Or do you have to pay for it like monthly/annually?

Yes, my company bought 5 "Station Control" licenses, but with those licenses you can't run more than 15 jobs and you can't send torque/angle value using open protocol. You need to use "Critical Control" license. The script works perfectly with these licenses.

closed 2024-09-22 by Leonel7