Rickedb/OpenProtocolInterpreter #119: MidInterpreter.GetMessageTemplate(int mid) is failing when calling Parse

tjarnagin, 2025-09-16:
We are using the latest version - 6.1

When the controller send a response, in this case MID 2, an exception is thrown:

Exception has been thrown by the target of an invocation. (mscorlib)

------------------------------
Program Location (Client side):

   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes, StackCrawlMark& stackMark)
   at System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
   at System.Activator.CreateInstance(Type type, Object[] args)
   at OpenProtocolInterpreter.MidInterpreter.<>c__DisplayClass13_0`1.<UseTemplate>b__0()
   at System.Lazy`1.CreateValue()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Lazy`1.get_Value()
   at OpenProtocolInterpreter.MidInterpreter.<>c__DisplayClass15_0.<GetMessageTemplate>b__0(Lazy`1 x)
   at System.Linq.Enumerable.FirstOrDefault[TSource](IEnumerable`1 source, Func`2 predicate)
   at OpenProtocolInterpreter.MidInterpreter.GetMessageTemplate(Int32 mid)
   at OpenProtocolInterpreter.MidInterpreter.Parse(Byte[] package)
   at DesoutterActivityViewModel.OnDataReceived(Byte[] message)

------------------------------
ADDITIONAL INFORMATION:

Collection was modified; enumeration operation may not execute. (mscorlib)

------------------------------
Program Location (Client side):

   at System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   at System.Collections.Generic.Dictionary`2.Enumerator.MoveNext()
   at System.Linq.Enumerable.WhereEnumerableIterator`1.MoveNext()
   at OpenProtocolInterpreter.Messages.MessagesTemplate.FilterSelectedMids(IEnumerable`1 mids)
   at OpenProtocolInterpreter.Messages.MessagesTemplate.FilterSelectedMids(IEnumerable`1 mids)
   at OpenProtocolInterpreter.Communication.CommunicationMessages..ctor(IEnumerable`1 selectedMids)

tjarnagin, 2025-09-17:
The exception is coming from the following line in GetMessageTemplate(int mid):

var lazy = _messagesTemplates.Values.FirstOrDefault(x => x.Value.IsAssignableTo(mid));

Rickedb, 2025-09-18:
I did reach the same point, however, its strange because the exception should happen only when someone modifies the list during the FirstOrDefault operation.
However, this is only modified when executing `UseTemplate()` via `MidInterpreterMessagesExtensions` extension methods.

How are you registering the mids you want to use/parse?

tjarnagin, 2025-09-18:
Okay, I found that commit 435f7295: #108  Mid0902 tests and implementation caused the issue.

Class MessageTemplate: FilterSelectedMids(IEnumerable<MidCompiledInstance> mids) had ToList() removed so now it is removing from the same collection it is iterating over.

I altered this and the exception is gone.

tjarnagin, 2025-09-18:
> CAUTION: THIS EMAIL ORIGINATED OUTSIDE OF AEGIS

I posted our construction of the interpreter which is done in our class constructor. The error happens after connecting to the controller and the first MID response from the controller.

No matter when that method is accessed you can't alter a collection while iterating it.

Tim Jarnagin